xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface #11: File upload error

saloid, 2022-01-11:
Hi. I tried to update to your firmware from release using web page (previosly there was original firmware). I uploaded webView file first and it was succesfull, but when I tried to upload main firmware I got "File error". And now I can't update both - getting same error. Also I can't downgrade to [original firmware](https://github.com/xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface/raw/master/firmware_release/MksWifi_WebView.bin)
[image]
[image]

saloid, 2022-01-11:
I still can control my printer in "MKS cloud" app

xreef, 2022-01-11:
Hi Saloid,
for now, download previous version and put It in the sd, then restart It.
I check if there is a problem on current version.
Bye Renzo

saloid, 2022-01-11:
@xreef I put both files on SD card and succesfully updated. Thanks for your support)

xreef, 2022-01-11:
Sometime if wifi isn't strength enough there can be transfer problem and artifact.
Next time, try to stay close to the wifi router with the printer.
Bye Renzo

closed 2022-01-11 by xreef