xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface #13: Can't flash into FBG5. Update resets wifi settings

Regressor, 2022-01-19:
Hi, I tried to flash this and there is a problem. After pressing update button (webview or firmware) there is a ~30seconds delay and then error "server reset connection". Then printer is not available by ip and display show its wifi settings gone. After joining wifi network I can see default update web page.

Regressor, 2022-01-20:
Flashed it using sd card and ftp. Works.

WonderCRM, 2022-01-20:
> Perfect! Probably low strength on wifi connection generate that problem.
> Thanks for your feedback.

The problem is something else, I also could not flash the option from the video, I had to invent it.

xreef, 2022-01-20:
I upload the latest version on my flying Bear ghost 5 without problem.
I had that problem when i upload It to the mks robin board that i use for testing, and the wifi was more far.
But i do some other upload tests.
Bye Renzo

WonderCRM, 2022-01-20:
> I upload the latest version on my flying Bear ghost 5 without problem.
> I had that problem when i upload It to the mks robin board that i use for testing, and the wifi was more far.
> But i do some other upload tests.

It turned out to update only through a memory card. Web files via ftp server after adding it to MksWiFi.

Regressor, 2022-01-21:
> Perfect! Probably low strength on wifi connection generate that problem. Thanks for your feedback.

Every time I tried to flash it using web mks wifi booted without wifi settings and I had to rejoin it to wifi again and again. It looks like last fbg5 mks wifi updates changed something and this prevents flashing your fw.

xreef, 2022-01-21:
Try to use the original firmware that I share on the article, then flash my firmware like explained on the article.
Give me a feedback..

closed 2022-01-29 by xreef