xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface #14: Compiling the frontend

gerwant, 2022-01-27:
Hello, I have tried to compile .bin files from the source code and upload them to my printer. I have successfully compiled one of them (firmware) but I can't see an option to compile second .bin (that one with web interface). Could anyone help me with that? Thanks in advance!

xreef, 2022-01-27:
Hi @gerwant,
you must generate spiff bin file, i write a complete guide about firmware generation

https://www.mischianti.org/category/tutorial/esp8266-tutorial/esp8266-firmware-management-tutorial/

Bye Renzo

xreef, 2022-01-27:
You can find the articles about msk wifi here

https://www.mischianti.org/category/project/web-interface-beeprint-for-mks-wifi/

Bye Renzo

gerwant, 2022-01-28:
Okay, thank you, I will try to upload it that way. But anyway, how can I change something in the frontend? It looks like the content of "data" directory is an already built app and changing anything there is quite problematic. Do you share the source code of the frontend somewhere? Thanks again!

xreef, 2022-01-29:
No sorry, I'd like to follow the develop personally (check also the license), i'd like to find some collaborator if you are interested.
The application It's React, Redux web application.

Bye Renzo

closed 2022-01-29 by xreef