xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface #16: Mks Robin Pro v1. 0

Moz72, 2022-02-19:
Mks Robin Pro. Web ui starts but no temp nothing is showing or updating.

xreef, 2022-02-21:
Can you send me a link of the board and the firmware you had installed?
Bye Renzo

Moz72, 2022-02-21:
Ich habe es hin bekommen, aber frag mich nicht wie🙈MeinDrucker ist ein Jgmaker Artist-d Pro. Der ein Mks robin pro v1.0 verbaut hat. Installiert habe ich aber ein firmware vom mks nano35 board plus die wifi.bin und dann ging alles.
Super Arbeit von dir👍👍👍👍👍

xreef, 2022-02-21:
Hehehhehe... good!
Bye Renzo

closed 2022-02-21 by xreef

Moz72, 2022-02-24:
Es hat leider doch nicht geklappt😢
Habe dir eine Mail geschickt, ist die angekommen?
Gruß Marcus

xreef, 2022-02-25:
Hi @Moz72,
I don't have the board so I can't do debug or other.
Sorry Renzo