xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface #18: Uploading Files

rwoldberg, 2022-03-25:
I am having upload issues.  Most of the time the file upload does not complete.  I put some debug statments in the wifi-module, and it appears that it never reveived the lastFragment flag.  I think this is an issue in wifi firmware, however I am unable to compile it as the esp8266 core can not be downloaded from github as it appears there is a virus in one of the files.

xreef, 2022-03-25:
Hi rwoldberg,
try to put your printer near to the router, probably the RSSI is not good.
Probably is the web part, some antivirus consider JS file malicious.
Bye Renzo

xreef, 2022-03-30:
Hi,
I don't use that core, I use a standard one for my project.

I can't replicate the problem.

Bye renzo

closed 2022-03-30 by xreef