xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface #19: [Feature Request] GCODE execution Output

nexius76, 2022-04-08:
C'è un modo per recuperare l'output del codice g inviato che non implica funzioni di movimento? Vorrei prendere la configurazione del firmware inviando M501 ma non so dove recuperare l'output.

xreef, 2022-04-27:
Scusami, avevo perso questo issue.
Tramite il modulo MKS WiFi ci sono solo un set minimo di comandi con output, per gli altri c'è disponibile solel'esito.
Ciao Renzo

nexius76, 2022-04-27:
Ah ok. E secondo sarebbe possibile avere l'output dell'esecuzione di M501? Sarebbe molto comodo poter avere conferma delle impostazioni attive sul firmware senza dover per forza attaccare un cavo seriale.

Grazie comunque! Ottimo lavoro.

----------------------------------------

*Da: *Renzo Mischianti ***@***.***>
*A: *xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface ***@***.***>
*CC: *nexius76 ***@***.***>; Author ***@***.***>
*Data: *27 apr 2022 14:51:30
*Oggetto: *Re: [xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface] [Feature Request] GCODE execution Output (Issue #19)