xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface #20: How do I modify the UI of the web?

CLP58, 2022-04-25:
Hi Renzo Mischianti
I would like to modify it to suit laser engraving use. I am a novice and not very familiar with Arduino. Can I build it using VScode? Any advice and tutorials? Many thanks!

CLP58, 2022-04-25:
I tried to compile using Arduino and got the following error.

The project is being compiled...
"C:\\Users\\yoy\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\python3\\3.7.2-post1/python3" -I "C:\\Users\\yoy\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2/tools/signing.py" --mode header --publickey "C:\\Users\\yoy\\Desktop\\WEB\\MKS_WIFI_upgrade_with_BeePrint_web_interface-master\\MKS_WIFI_source/public.key" --out "C:\\Users\\yoy\\AppData\\Local\\Temp\\arduino_build_187402/core/Updater_Signing.h"
"C:\\Users\\yoy\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE "-IC:\\Users\\yoy\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2/tools/sdk/include" "-IC:\\Users\\yoy\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2/tools/sdk/lwip2/include" "-IC:\\Users\\yoy\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\yoy\\AppData\\Local\\Temp\\arduino_build_187402/core" -c -w -Werror=return-type -Os -g -free -fipa-pta -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -MMD -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10819 -DARDUINO_ESP8266_NODEMCU_ESP12E -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU_ESP12E\"" -DLED_BUILTIN=2 -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\yoy\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2\\cores\\esp8266" "-IC:\\Users\\yoy\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2\\variants\\nodemcu" "-IC:\\Users\\yoy\\Documents\\Arduino\\libraries\\ArduinoJson\\src" "-IC:\\Users\\yoy\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2\\libraries\\ESP8266WiFi\\src" "-IC:\\Users\\yoy\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2\\libraries\\ESP8266mDNS\\src" "-IC:\\Users\\yoy\\Documents\\Arduino\\libraries\\eFLL" "-IC:\\Users\\yoy\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2\\libraries\\EEPROM" "-IC:\\Users\\yoy\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2\\libraries\\ESP8266HTTPClient\\src" "-IC:\\Users\\yoy\\Documents\\Arduino\\libraries\\WebSockets\\src" "-IC:\\Users\\yoy\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2\\libraries\\Hash\\src" "C:\\Users\\yoy\\AppData\\Local\\Temp\\arduino_build_187402\\sketch\\MKS_WIFI_source.ino.cpp" -o "C:\\Users\\yoy\\AppData\\Local\\Temp\\arduino_build_187402\\sketch\\MKS_WIFI_source.ino.cpp.o"
In file included from C:\Users\yoy\Desktop\WEB\MKS_WIFI_upgrade_with_BeePrint_web_interface-master\MKS_WIFI_source\MKS_WIFI_source.ino:21:
C:\Users\yoy\Documents\Arduino\libraries\eFLL/RepRapWebServer.h: In instantiation of 'size_t RepRapWebServer::streamFile(T&, const String&) [with T = fs::File; size_t = unsigned int]':
C:\Users\yoy\Desktop\WEB\MKS_WIFI_upgrade_with_BeePrint_web_interface-master\MKS_WIFI_source\MKS_WIFI_source.ino:603:60:   required from here
C:\Users\yoy\Documents\Arduino\libraries\eFLL/RepRapWebServer.h:140:30: error: no matching function for call to 'WiFiClient::write(fs::File&, int)'
  140 |   return _currentClient.write(file, HTTP_DOWNLOAD_UNIT_SIZE);
      |          ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\yoy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:39,
                 from C:\Users\yoy\Desktop\WEB\MKS_WIFI_upgrade_with_BeePrint_web_interface-master\MKS_WIFI_source\MKS_WIFI_source.ino:18:
C:\Users\yoy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src/WiFiClient.h:59:18: note: candidate: 'virtual size_t WiFiClient::write(uint8_t)'
   59 |   virtual size_t write(uint8_t) override;
      |                  ^~~~~
C:\Users\yoy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src/WiFiClient.h:59:18: note:   candidate expects 1 argument, 2 provided
C:\Users\yoy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src/WiFiClient.h:60:18: note: candidate: 'virtual size_t WiFiClient::write(const uint8_t*, size_t)'
   60 |   virtual size_t write(const uint8_t *buf, size_t size) override;
      |                  ^~~~~
C:\Users\yoy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src/WiFiClient.h:60:39: note:   no known conversion for argument 1 from 'fs::File' to 'const uint8_t*' {aka 'const unsigned char*'}
   60 |   virtual size_t write(const uint8_t *buf, size_t size) override;
      |                        ~~~~~~~~~~~~~~~^~~
C:\Users\yoy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src/WiFiClient.h:62:10: note: candidate: 'size_t WiFiClient::write(Stream&)'
   62 |   size_t write(Stream& stream) [[ deprecated("use stream.sendHow(client...)") ]];
      |          ^~~~~
C:\Users\yoy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src/WiFiClient.h:62:10: note:   candidate expects 1 argument, 2 provided
[... cut ...]
Use the 1.2 version of the library ESP8266HTTPClient in the folder: C:\Users\yoy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ ESP8266HTTPClient 
Use the 2.3.6 version of the library WebSockets in the folder: C:\Users\yoy\Documents\Arduino\libraries\WebSockets 
Use version 1.0 of the library Hash in the folder: C:\Users\yoy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\Hash 
exit status 1
Error while compiling for development board NodeMCU 1.0 (ESP-12E Module).

Translated with www.DeepL.com/Translator (free version)

xreef, 2022-04-25:
https://www.mischianti.org/forums/topic/how-do-i-modify-the-ui-of-the-web/

closed 2022-04-25 by xreef