xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface #22: Error caused by Chinese path in SD card

thk233, 2022-07-22:
I am a Chinese user. In the process of using, I noticed that there seems to be no Chinese path in the SD card directory, which will lead to frequent disconnection of printers in the storage interface. Will there be access support for Chinese interface and Chinese path in the future?

The above is using translation machine translation. I'm sorry if the abnormal translation causes offence

guumeyer, 2022-07-22:
I noticed longer name or character with accent. Result on failure during the file transmission to the printer

xreef, 2022-07-25:
Hi all,
for the longer name, I add the management (not released I think), for the special character It's strange, I must check the encoding on the web interface, but Cyrillic work well.
Bye Renzo