xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface #23: Add light and/or PSU control

meHaSQUE, 2022-08-08:
I have been dreaming for a long time to be able to turn on the printer remotely, as well as control the backlight.
Can you please add a couple of buttons to the interface, and I will try to add these functions to the back end??

I see the implementation like this:
connect 2-3 relays to free pins on esp.
- One relay will turn off the main power supply.
- The second relay (and the third?) will turn on the backlight (which is powered by the printer's built-in 24v power supply).

It will be great if I can control two lights from the web interface, such as a switch or a slider, since I have a separate one for recording time-lapses, but it is too bright to be on all the time.

In order for the circuit to continue working when the 24v power supply is turned off, I suggest that the main motherboard (together with the ESP) can be connected to a separate 5v power supply, for example from an old phone or tablet, then the ESP will remain online even if the main power supply is turned off and you can will turn it on remotely over the network.

xreef, 2022-08-10:
Hi @meHaSQUE,
I think that development in the past, the solution can be to use the esp32-cam as the main controller (powered with 5v separately).

The first idea was to send an email with the picture of the finished print and then power off the printer.

But I have few time, if you help me we start to develop all the features on UI without a problem.
Bye Renzo

meHaSQUE, 2022-08-13:
Hi @xreef ,

Looks like i can add end point to control power/light. The source code look messy but understandable.
Maybe I will be able to add PWM as well.

As additional feature I'm thinking about adding this controls to Home Assistant.

Does it look like as a first step?

xreef, 2022-08-16:
Hi,
I think the best solution is to not overweight the code on WifiModule, I get a lot of problems with the memory when adding WebSocket, the original implementation is quite hungry for resources.

I think can be a good idea to add only client rest calls with a minimal configuration.

But if you want to follow another way I support you.

Bye Renzo