xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface #28: Portuguese translate

RonaldoRisso, 2023-01-31:
hi, i added the portuguese translation in the file
I couldn't compile here

https://drive.google.com/drive/folders/1slwaNuX-UHMA1ACqXRykcxYB_toKId8X

RonaldoRisso, 2023-01-31:
corrected and reuploaded data.json

xreef, 2023-02-03:
added

closed 2023-02-03 by xreef

xreef, 2023-02-04:
Give me feedback on the lastest version with your translation.
Thanks Renzo

RonaldoRisso, 2023-02-04:
> Give me feedback on the lastest version with your translation. Thanks Renzo

https://drive.google.com/file/d/1H2cBc8K6l7H9iZ70LDcCJkRojqNQWXzc/view?usp=share_link
https://drive.google.com/file/d/1E1V4yfeqq_G3Pqw2z_jBeRl0zTQ3aOCe/view?usp=share_link
Works fine, but i adjusted the texts that were out of layout and updated the file
new data.json: https://drive.google.com/file/d/1PWMIqHFT2ilJSFrIpnbhZnKBipGTDl4l/view?usp=share_link

Tk´s

RonaldoRisso, 2023-02-06:
waiting for a 1.7 or 16.5 version... :)

RonaldoRisso, 2023-02-08:
version 1.6.1 works perfectly, the portuguese translation It is ok
Tk´s again !!!

xreef, 2023-02-09:
Thanks to you for the support.
Bye Renzo

RonaldoRisso, 2023-02-17:
Latest with camera
https://drive.google.com/file/d/1zJO1m-iflEajpjJwEHToxWcRZJ5y8qT4/view?usp=share_link

xreef, 2023-02-17:
uuuu... thanks for sharing.. probably i add the image to the site.
Bye Renzo