xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface #29: Marlin support

eduard-sukharev, 2023-02-17:
It would be cool to have Marlin support alongside with RRF support.
I know, this firmware is basically a modded MKS WiFi firmware which only supports RRF as far as I can tell, but having a config parameter to build for Marlin support would be just great.
Any plans on that? Any hints on where to start to get things running on Marlin?

barkh, 2023-02-27:
MKS wifi supports Marlin out of box .... ESP3D is working on both RRF and Marlin too. Only work you must do is use different G and M codes for some functions.

xreef, 2023-02-28:
Hi all,
I don't have a marlin on my developer board, I also think that the change can be very simple, but I don't know what I must do.
I accept help for this task.
Bye Renzo

barkh, 2023-02-28:
I do not do react and I think important part with GCODEs is missing in github. I have MKS Robin Nano 1.2 and MKS WiFi on actual Marlin. 

If you publish web part with GCODEs or send me used GCODEs and their parameters and expected response, I can send you Marlin eqvivalents. For future support of different firmwares it would be best to have some configuration table with GCODES for each firmware.

If you want to extend support to BTT WiFi, I have BTT SKR 2 on Marlin and BTT WiFi and BTT TFT with BTT WiFi.

xreef, 2023-02-28:
Hi barkh,
the GCode "reading" part is here
https://github.com/xreef/beeprint_3d_printer_web_interface/blob/master/src/redux/logic/realtimeUpdate.jsx#L102-L177

the command to send is here
https://github.com/xreef/beeprint_3d_printer_web_interface/blob/master/src/redux/additions/commands.js

I think you need these. For other info, I'm here.

Bye Renzo

barkh, 2023-02-28:
For what firmware it was written? I did not find any documentation on FlyingGhost. Some small hits saying , it is based on Marlin? But I look on both Marlin and RepRap and some GCODEs are wrong

multipleStatRequest
M997 - firmware upgrade??? on both marlin and reprap https://reprap.org/wiki/G-code#M997:_Perform_in-application_firmware_update, i did not know, what it supposed to do
M994 - only for Marlin -  load a backup from SD to SPI Flash, i did not know, what it supposed to do
M992 - i did not find this gcode anywhere, i did not know, what it supposed to do

requestFileList
M20 - probably P parameter is missing there (inrtended commad is probably M20 P"1:/")
Marlin lacks P parameter, but introduces L (long filenames) and T (timestamp) parameters, which should be used

printStop
M26 - it is set position on both RepRap and Marlin, without S parameter useles

It is just sample, withou documentation on GCODEs on command.js i can not help.

xreef, 2023-02-28:
Hi,
you can find the GCode commands here
https://www.mischianti.org/2021/11/24/mks-wifi-for-makerbase-robin-communication-protocol-and-cura-plugin-3/#MKS_WiFi_protocol

But they are the MKS_WIFI commands that are translated and returned from the MKS_WIFI firmware so
https://github.com/xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface/blob/master/MKS_WIFI_upgrade_with_BeePrint_web_interface.ino#L1560-L1656

The standard firmware is the makerbase one 
https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X

Bye Renzo

eduard-sukharev, 2023-02-28:
I think RepRap wiki is outdated:
Marlin has `M997` support
https://marlinfw.org/docs/gcode/M997.html

To report printing time and percentage (`M992` and `M27` in code above) Marlin would use `M73` https://marlinfw.org/docs/gcode/M073.html or `M31` https://marlinfw.org/docs/gcode/M031.html

The Stop SD print `M26` maps to `M524` in Marlin https://marlinfw.org/docs/gcode/M524.html

The resetPrinter sequence is kinda questionable, I'd use `M997` for that (in most cases AFAIK it just power-cycles the board).

And there's also two cool-downs - `cooldown` to shut heaters down and `coolDown` to shut heaters down and turn fan on full throttle.

barkh, 2023-03-01:
I strongly recomend to go to Marlin - color UI, not LVGL (for nano 2.0 it should be #define MKS_TS35_V2_0 (you probably have TS35-R, but as I know it is same only without encoder) and #define TFT_COLOR_UI) - LVGL is similar to Makerbase FW - important futures and settings is mising there - you must compile FW, if you want to change something. I have sensorless homing and TMC UART (you need 4 wires on nano 1.2), filament runout, automatic bed leveling, junction deviation and MPCTEMP - works like dream. For commands:

preHeatTool: same
multipleStatRequest:
- M105 - can be used, but M155 is recomended to use for Marlin - but you must decode printer capatibilities for it (M115) to know M155 is supported (line AUTOREPORT_TEMP:1)
- M997 - it is dangerous for Marlin - firmware update. Most likely use M31 - returns "Print time: xxx",  used with M75, M76 and M77
- M994 - M27 C for filename (one or two strings, first is DOS name, optional second is long name). For size use M27, returns "SD printing byte 123/12345". You can use M27 Sx, if M155 returns AUTOREPORT_SD_STATUS:1
- M992 - see M31
- M27 - different format, returns "SD printing byte 123/12345"
requestFileList:
- M20 - different format, use M20 L (https://marlinfw.org/docs/gcode/M020.html)
selectAndPrint:
- M23 - same
- M24 - same (if you do not want to solve power recovery)
printSelectFile:
- M23 - probably wrong command or wrong function name, should be M24, M23 is select file
printResumeStart: same
printPause: same
printStop:
- M26 - different, use M524
deleteFile:
- M30 - different formatm only M30 filename (without 1:)
setFanSpeed: same
unlock: same
moveAxes: same
- there is probably mistake in G0 command - ".0" should not be here, it is already added yA and zA
extrudeFilament: same
homingXY: same
homingZ: same
homingAll: same
resetPrinter: same
coolDown: same
- i thing fan speed should be left unmodified, it is part cooling fan, not extruder cooling fan
unlockMotors: same

You can not use M73, M73 is for slicer to put estimate times into GCODE, not to read them. There will probably be support for M408, but they are pushing it for 5 years. It is in bugfix branch of marlin and it is for panel due only - so it means custom firmware to remove if. I put comment to their issue.

Im going to configure Marlin to ESP3D and I write changes to be done to marlin to get it work.

eduard-sukharev, 2023-03-01:
My initial idea was to get Marlin support in BeePrint on boards with MKS WiFi, regardless of whether there is a screen or anything else. I don't see how using one screen or another should affect the BeePrint support — I only have MKS Mini 12864 v3 LCD screen and it doesn't support neither Color UI, nor LVGL.

The idea here is that I want to have web-ui with SD card upload and print feature so that I don't have to
- pick SD from my printer
- run with it across the room
- insert into card reader
- mount it
- save gcode to it
- unmount
- run across room
- insert into printer
- enter print menu
- pick a file
- confirm print start

What I want is to:
- open webui
- upload gcode
- press print

barkh, 2023-03-10:
Today I go through marlin and ESP3D. Result is, that Marlin has MKS wifi, but only for LVGL - MKS hardcoded ESP WIFI into LVGL GUI ...
Important files are wifi_module.(c/h) - handles comunication with ESP and wifi_upload.(c/h) - handles firmware upgrade of ESP. Somebody, who programs in C++ can probably liberate this from LVGL gui restriction, it seems, that using LVGL is only shortcut to not implement gui into DOGM/TFT_COLOR_GUI.

eduard-sukharev, 2023-03-10:
Yep, MKS WiFi module is hard-tied to LVGL UI in Marlin and it's core protocol implementation is intertwined with GCode parsing and UI updating in `wifi_module.cpp`. Luc from ESP3D has a [FeatureRequest in ESP3D](https://github.com/luc-github/ESP3D/issues/575) to refactor out MKS protocol to a discrete library, but it's only an outline of his plans and he has no capacity of creating such library so far.

That being said, Marlin support would require fast enough file upload protocol support on both sides and that's outside of the scope of this request, so closing this issue.

closed 2023-03-10 by eduard-sukharev