xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface #30: Printer not connected

array81, 2023-02-23:
I bought a MKS Robin WIFI, installed on my FLSun SR (MKS Nano 3), set my WIFI data and upload your firmware.
When I try to access to web interface I can see your interface and, on start, the messagge "print connected", however the printer is not connected in fact in all the webserver page (for example temparature, print, ...) I see "not connected" label.

xreef, 2023-02-28:
Retry the installation process.
Then try to install BeePrint 1.5 and MKS_WIFI 1.2.
Bye Renzo

alexanderphoenix, 2023-05-17:
Hi @xreef ,
I am trying to get your great firmware working on my FLsun Q5 however I get the same as @array81.
As you suggested I have tried both `F1.3_B1.6.1` and `F1.2_B1.5` like you suggested above, but all result in the `not connected` message.
Any further tips on how to resolve this please?
The FLsun Q5 I have run on a `MKS Robin Nano V1.3` if that helps.
Thank you very much

Carontes, 2023-11-25:
Same problem here. I got MKS Robin nano 1.2 and MKS WiFi module, I installed the latest version of both webview and firmware. I can connect via web after my router gave it an IP address, but the printer appears as not connected, even the wifi signal on the top-right corner is gray. Any suggestion?

xreef, 2023-11-28:
Hi all,
It's possible that you don't use the original firmware?
People also test the device on Ghost6 (with the same MB version) and works.
Bye Renzo

Carontes, 2023-11-28:
I use custom Marlin 2.1.x bugfix, but the original firmware of my printer is always Marlin but 1.0.2

xreef, 2023-11-28:
We continue on the forum, and I need the correct baud rate perimeter to make the correct choice.
[Topic about this](https://mischianti.org/forums/topic/printer-not-connected/)
Bye RM