xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface #31: Remove WIFI signal debug print

thecrow159, 2023-04-11:
Hi,
I'm using Repetier-Host to connect with the printer using WiFi.
I always get WIFI: -78 (the signal strength) printed on the log windows. I think that it becomes from the esp and the serial connection emulated with TCP/IP connection. Is it possibile to remove those prints?

Thank you

xreef, 2023-04-16:
Hi,
I use that information to show on the web interface

[image]

There is any problem with some devices?

thecrow159, 2023-04-16:
Hi Renzo,
thank you for the answer and the great project. It isn't causing any issue. It's just frustrating having those prints in the log console while debugging  gcode so I was asking if there was a way to stop them, even from time to time.

Thank you for your time,
Emiliano

xreef, 2023-04-19:
Hi Emiliano,
sorry for the late response, I.m going to add a define to remove It, and if you need It I generate a specified bin file.
Bye Renzo

thecrow159, 2023-04-21:
> Hi Emiliano, sorry for the late response, I.m going to add a define to remove It, and if you need It I generate a specified bin file. Bye Renzo

Hi Renzo,
thank you. It would be great!

Emiliano