xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface #33: Cant Compile

Stewie-040882, 2023-06-22:
In file included from C:\Users\..\Downloads\MKS_WIFI_upgrade_with_BeePrint_web_interface-2.7.4_esp8266_core\MKS_WIFI_source\RepRapWebServer.cpp:28:
C:\Users\..\Downloads\MKS_WIFI_upgrade_with_BeePrint_web_interface-2.7.4_esp8266_core\MKS_WIFI_source\RepRapWebServer.h: In instantiation of 'size_t RepRapWebServer::streamFile(T&, const String&) [with T = fs::File; size_t = unsigned int]':
C:\Users\..\Downloads\MKS_WIFI_upgrade_with_BeePrint_web_interface-2.7.4_esp8266_core\MKS_WIFI_source\RequestHandlersImpl.h:180:41:   required from here
C:\Users\..\Downloads\MKS_WIFI_upgrade_with_BeePrint_web_interface-2.7.4_esp8266_core\MKS_WIFI_source\RepRapWebServer.h:140:30: error: no matching function for call to 'WiFiClient::write(fs::File&, int)'
  140 |   return _currentClient.write(file, HTTP_DOWNLOAD_UNIT_SIZE);
      |          ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\..\Downloads\MKS_WIFI_upgrade_with_BeePrint_web_interface-2.7.4_esp8266_core\MKS_WIFI_source\RepRapWebServer.cpp:27:
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WiFi\src/WiFiClient.h:71:18: note: candidate: 'virtual size_t WiFiClient::write(uint8_t)'
   71 |   virtual size_t write(uint8_t) override;
      |                  ^~~~~
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WiFi\src/WiFiClient.h:71:18: note:   candidate expects 1 argument, 2 provided
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WiFi\src/WiFiClient.h:72:18: note: candidate: 'virtual size_t WiFiClient::write(const uint8_t*, size_t)'
   72 |   virtual size_t write(const uint8_t *buf, size_t size) override;
      |                  ^~~~~
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WiFi\src/WiFiClient.h:72:39: note:   no known conversion for argument 1 from 'fs::File' to 'const uint8_t*' {aka 'const unsigned char*'}
   72 |   virtual size_t write(const uint8_t *buf, size_t size) override;
      |                        ~~~~~~~~~~~~~~~^~~
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WiFi\src/WiFiClient.h:75:10: note: candidate: 'size_t WiFiClient::write(Stream&)'
   75 |   size_t write(Stream& stream);
      |          ^~~~~
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WiFi\src/WiFiClient.h:75:10: note:   candidate expects 1 argument, 2 provided
In file included from C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Stream.h:27,
                 from C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/HardwareSerial.h:32,
                 from C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Arduino.h:303,
                 from C:\Users\..\Downloads\MKS_WIFI_upgrade_with_BeePrint_web_interface-2.7.4_esp8266_core\MKS_WIFI_source\RepRapWebServer.cpp:24:
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:76:23: note: candidate: 'size_t Print::write(int8_t)'
   76 |         inline size_t write(int8_t c) { return write((uint8_t) c); }
      |                       ^~~~~
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:76:23: note:   candidate expects 1 argument, 2 provided
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:75:23: note: candidate: 'size_t Print::write(char)'
   75 |         inline size_t write(char c) { return write((uint8_t) c); }
      |                       ^~~~~
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:75:23: note:   candidate expects 1 argument, 2 provided
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:73:23: note: candidate: 'size_t Print::write(long long unsigned int)'
   73 |         inline size_t write(unsigned long long t) { return write((uint8_t)t); }
      |                       ^~~~~
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:73:23: note:   candidate expects 1 argument, 2 provided
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:72:23: note: candidate: 'size_t Print::write(long long int)'
   72 |         inline size_t write(long long t) { return write((uint8_t)t); }
      |                       ^~~~~
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:72:23: note:   candidate expects 1 argument, 2 provided
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:71:23: note: candidate: 'size_t Print::write(long unsigned int)'
   71 |         inline size_t write(unsigned long t) { return write((uint8_t)t); }
      |                       ^~~~~
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:71:23: note:   candidate expects 1 argument, 2 provided
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:70:23: note: candidate: 'size_t Print::write(long int)'
   70 |         inline size_t write(long t) { return write((uint8_t)t); }
      |                       ^~~~~
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:70:23: note:   candidate expects 1 argument, 2 provided
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:69:23: note: candidate: 'size_t Print::write(unsigned int)'
   69 |         inline size_t write(unsigned int t) { return write((uint8_t)t); }
      |                       ^~~~~
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:69:23: note:   candidate expects 1 argument, 2 provided
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:68:23: note: candidate: 'size_t Print::write(int)'
   68 |         inline size_t write(int t) { return write((uint8_t)t); }
      |                       ^~~~~
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:68:23: note:   candidate expects 1 argument, 2 provided
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:67:23: note: candidate: 'size_t Print::write(short unsigned int)'
   67 |         inline size_t write(unsigned short t) { return write((uint8_t)t); }
      |                       ^~~~~
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:67:23: note:   candidate expects 1 argument, 2 provided
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:66:23: note: candidate: 'size_t Print::write(short int)'
   66 |         inline size_t write(short t) { return write((uint8_t)t); }
      |                       ^~~~~
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:66:23: note:   candidate expects 1 argument, 2 provided
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:62:16: note: candidate: 'size_t Print::write(const char*, size_t)'
   62 |         size_t write(const char *buffer, size_t size) {
      |                ^~~~~
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:62:34: note:   no known conversion for argument 1 from 'fs::File' to 'const char*'
   62 |         size_t write(const char *buffer, size_t size) {
[... 92 lines cut ...]
      |                       ^~~~~
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:68:23: note:   candidate expects 1 argument, 2 provided
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:67:23: note: candidate: 'size_t Print::write(short unsigned int)'
   67 |         inline size_t write(unsigned short t) { return write((uint8_t)t); }
      |                       ^~~~~
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:67:23: note:   candidate expects 1 argument, 2 provided
[... cut ...]
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:56:16: note: candidate: 'size_t Print::write(const char*)'
   56 |         size_t write(const char *str) {
      |                ^~~~~
C:\Users\..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/Print.h:56:16: note:   candidate expects 1 argument, 2 provided
cc1plus.exe: some warnings being treated as errors

exit status 1

Compilation error: no matching function for call to 'WiFiClient::write(fs::File&, int)'

xreef, 2023-06-23:
Thanks,
yes probably yes, but that are the original function from mks release, I'm going to check.
Bye Renzo

Stewie-040882, 2023-06-23:
im doing my best to learn, compile, and flash so that i can try to assist you. but ive tested around 4 compiles today but broke wifi connection each time, soim doing something wrong lol

xreef, 2023-06-24:
WIFImanager hasn't the management of the dns, so it can't work with the standard version.
Here the fixed version.
https://github.com/xreef/WiFiManager
Bye Renzo

Stewie-040882, 2023-06-24:
That's not the problem, I use "mks.bin" and connect to WiFi, I then upload "beeprint Web.bin", then upload "beeprint wifi.bin" everything working great.
  I compile new "beeprint wifi.bin" from code using arduino IDE, and wifi is broken "wifi starting up" message on printer.  Flash "beeprint wifi.bin" everything's great again. 
   I need to learn coding and compiling. Your code is great 👍

xreef, 2023-06-26:
Hi Stewie,
did you check this article?
https://mischianti.org/2021/11/21/mks-wifi-for-makerbase-robin-pcb-and-how-to-compile-upload-firmware-2/
Here I added all steps to do the correct compilation (with the library I used).
Bye Renzo

closed 2023-06-26 by xreef