xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface #37: Printing from File List box does not work

bobafetthotmail, 2023-10-21:
Hello, I recently installed BeePrint 1.6.1 on my FliyngbearGhost 5 with stock firmware 5.91.

I have some gcode files created in the past that have been uploaded or copied to SD card.

I connect to web interface and I try to launch a print from the File List box I pinned to Home page, clicking on the "printer" icon of the file. I used Firefox, Chromium and Falkon browsers to repeat the test. Also restarted the printer with the power switch.

I see that the status goes to Printing for a few seconds but then returns Idle. I don't see any change on the touchscreen of the printer (when a print is started it will show printing information as most printers do)

If I upload a file using the "Upload and Print" button then it will start printing.

If I go to the printer and select the file to print from the touch screen, it also starts printing.

This does not seem related to file names. I can't print any file by clicking on the "printer" icon.

The buttons in the Printer Status box work fine, I can stop or pause or resume a print job.

xreef, 2023-10-22:
Hi,
try to format the SD card and try to use only the web interface.
Bye Renzo

bobafetthotmail, 2023-10-23:
reformatted SDcard (and tried with other sdcard).

Transferring files only using web interface and it is still has the same problems (some files with the names as in the other issue report are not loaded, printing files does not work from web interface)

bobafetthotmail, 2023-10-23:
I reflashed the firmware from https://mischianti.org/mks-wifi-for-makerbase-robin-firmware-upgrade-and-new-web-socket-features-4/

first the "pre-compiled original firmware" by placing the file in the sdcard and power off/on the printer
then the "pre-compiled firmware of the Mischianti" with web interface
    Take the MksWifi_WebView.bin and upload the Web View;
    Take MksWifi.bin and upload the firmware.

Then reformatted the SD card, and loaded some gcode files with web interface.

Now printing works for files I have loaded from web interface. It still cannot load the files with "bad" filenames.

If I copy the files with the "bad file names" to SD card from my PC, then it does again what I describe in the first post.

If I format the SD card again and loaded some gcode files with web interface.

And printing works again.

So this issue is connected to the https://github.com/xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface/issues/36 if there is a file with a bad name in the SD card it will not print too